jurobystricky/Netgear-A6210 #97: Fail to make on Kali Linux

Jimmydawnz, 2018-01-08:
`root@kali:~# git clone https://github.com/jurobystricky/Netgear-A6210
Cloning into 'Netgear-A6210'...
remote: Counting objects: 2436, done.
remote: Total 2436 (delta 0), reused 0 (delta 0), pack-reused 2436
Receiving objects: 100% (2436/2436), 3.71 MiB | 232.00 KiB/s, done.
Resolving deltas: 100% (1512/1512), done.
root@kali:~# cd Netgear-A6210
root@kali:~/Netgear-A6210# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.13.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
  CC [M]  /root/Netgear-A6210/os/linux/../../mgmt/mgmt_vht.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/vht.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/crypt_md5.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/crypt_sha2.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/crypt_hmac.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/crypt_aes.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/crypt_arc4.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/mlme.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_wep.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/action.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_data.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/rtmp_init.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/rtmp_init_inf.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_tkip.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_aes.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_sync.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/eeprom.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_sanity.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_info.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_cfg.o
  CC [M]  /root/Netgear-A6210/os/linux/../../common/cmm_wpa.o
In file included from /root/Netgear-A6210/include/rtmp_os.h:30:0,
                 from /root/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/Netgear-A6210/include/rt_config.h:34,
                 from /root/Netgear-A6210/os/linux/../../common/cmm_wpa.c:29:
/root/Netgear-A6210/os/linux/../../common/cmm_wpa.c: In function ‘ConstructEapolMsg.constprop’:
/root/Netgear-A6210/include/os/rt_linux.h:416:53: warning: argument 2 null where non-null expected [-Wnonnull]
 #define NdisMoveMemory(Destination, Source, Length) memmove(Destination, Source, Length)
                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/Netgear-A6210/os/linux/../../common/cmm_wpa.c:3742:3: note: in expansion of macro ‘NdisMoveMemory’
   NdisMoveMemory(&Key_Data[data_offset], GTK, gtk_len);
   ^~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/string.h:4:0,
                 from /usr/src/linux-headers-4.13.0-kali1-common/include/linux/string.h:18,
                 from /usr/src/linux-headers-4.13.0-kali1-common/include/linux/bitmap.h:8,
                 from /usr/src/linux-headers-4.13.0-kali1-common/include/linux/cpumask.h:11,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/cpumask.h:4,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/msr.h:10,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/processor.h:20,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/cpufeature.h:4,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/thread_info.h:52,
                 from /usr/src/linux-headers-4.13.0-kali1-common/include/linux/thread_info.h:37,
                 from /usr/src/linux-headers-4.13.0-kali1-common/arch/x86/include/asm/preempt.h:6,
                 from /usr/src/linux-headers-4.13.0-kali1-common/include/linux/preempt.h:80,
[... 44 lines cut ...]
  CC [M]  /root/Netgear-A6210/os/linux/../../ate/common/rt_ate.o
  CC [M]  /root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = CFG80211_OpsVirtualInfChg,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: note: (near initialization for ‘CFG80211_Ops.change_virtual_intf’)
/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = CFG80211_OpsVirtualInfAdd,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: note: (near initialization for ‘CFG80211_Ops.add_virtual_intf’)
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.13.0-kali1-common/scripts/Makefile.build:307: recipe for target '/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o' failed
make[4]: *** [/root/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
/usr/src/linux-headers-4.13.0-kali1-common/Makefile:1529: recipe for target '_module_/root/Netgear-A6210/os/linux' failed
make[3]: *** [_module_/root/Netgear-A6210/os/linux] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-kali1-amd64'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
`

HELP

kiriko31, 2018-02-05:
help cela ne fonctionne pas merci kali linux 4.14 erreur compilation make 2 erreurs exactement

21hertz, 2018-02-23:
Same here:

  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_scan.o
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .change_virtual_intf = CFG80211_OpsVirtualInfChg,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.c:2385:25: note: (near initialization for ‘CFG80211_Ops.change_virtual_intf’)
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .add_virtual_intf = CFG80211_OpsVirtualInfAdd,
                      ^~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.c:2386:22: note: (near initialization for ‘CFG80211_Ops.add_virtual_intf’)
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_rx.o
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_tx.o
cc1: some warnings being treated as errors
  CC [M]  /var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_inf.o
/usr/src/linux-headers-4.14.0-kali3-common/scripts/Makefile.build:325: recipe for target '/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.o' failed
make[4]: *** [/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211.o] Error 1
make[4]: *** Waiting for unfinished jobs....
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_inf.c: In function ‘RTMP_CFG80211_VirtualIF_Init’:
/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_inf.c:552:13: error: ‘struct net_device’ has no member named ‘destructor’; did you mean ‘priv_destructor’?
  new_dev_p->destructor = free_netdev;
             ^~~~~~~~~~
             priv_destructor
/usr/src/linux-headers-4.14.0-kali3-common/scripts/Makefile.build:325: recipe for target '/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_inf.o' failed
make[4]: *** [/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux/../../os/linux/cfg80211/cfg80211_inf.o] Error 1
make[4]: *** wait: No child processes.  Stop.
/usr/src/linux-headers-4.14.0-kali3-common/Makefile:1525: recipe for target '_module_/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux' failed
make[3]: *** [_module_/var/lib/dkms/netgear-a6210/2.5.0/build/os/linux] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.0-kali3-amd64'
Makefile:59: recipe for target 'debug' failed

timmy999989, 2018-04-09:
Hi,
i come this far, if i use switch -i but if i don't my output looks like this:
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.11-surface/build DBGFLAGS=-DDBG SUBDIRS=/root/wlan/Netgear-A6210/Netgear-A6210/os/linux modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.15.11-surface“ wird betreten
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../mgmt/mgmt_vht.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/vht.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/crypt_md5.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/crypt_sha2.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/crypt_hmac.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/crypt_aes.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/crypt_arc4.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/mlme.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_wep.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/action.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_data.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/rtmp_init.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/rtmp_init_inf.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_tkip.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_aes.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_sync.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/eeprom.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_sanity.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_info.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_cfg.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_wpa.o
In file included from /root/wlan/Netgear-A6210/Netgear-A6210/include/rtmp_os.h:30:0,
                 from /root/wlan/Netgear-A6210/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/wlan/Netgear-A6210/Netgear-A6210/include/rt_config.h:34,
                 from /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_wpa.c:29:
/root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_wpa.c: In function ‘ConstructEapolMsg.constprop’:
/root/wlan/Netgear-A6210/Netgear-A6210/include/os/rt_linux.h:416:53: warning: argument 2 null where non-null expected [-Wnonnull]
 #define NdisMoveMemory(Destination, Source, Length) memmove(Destination, Source, Length)
                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_wpa.c:3742:3: note: in expansion of macro ‘NdisMoveMemory’
   NdisMoveMemory(&Key_Data[data_offset], GTK, gtk_len);
   ^~~~~~~~~~~~~~
In file included from ./arch/x86/include/asm/string.h:5:0,
                 from ./include/linux/string.h:19,
                 from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /root/wlan/Netgear-A6210/Netgear-A6210/include/os/rt_linux.h:14,
[... 31 lines cut ...]
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/ba_action.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../mgmt/mgmt_ht.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_txbf.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/cmm_txbf_cal.o
  CC [M]  /root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../common/rt_os_util.o
[... cut ...]
scripts/Makefile.build:329: die Regel für Ziel „/root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../os/linux/rt_linux.o“ scheiterte
make[2]: *** [/root/wlan/Netgear-A6210/Netgear-A6210/os/linux/../../os/linux/rt_linux.o] Fehler 1
Makefile:1528: die Regel für Ziel „_module_/root/wlan/Netgear-A6210/Netgear-A6210/os/linux“ scheiterte
make[1]: *** [_module_/root/wlan/Netgear-A6210/Netgear-A6210/os/linux] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.15.11-surface“ wird verlassen
Makefile:59: die Regel für Ziel „debug“ scheiterte
make: *** [debug] Fehler 2