jurobystricky/Netgear-A6210 #98: __read_overflow2() error Ubuntu 17.10

dylan7581, 2018-01-10:
Hello,
I tried to make the Netgear-A6210 folder, but it does not work. Please help, this is make code execution : 
(ps : some line written in french)

`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/anonymed06/Documents/Netgear-A6210-master/os/linux/Makefile
make -C /lib/modules/4.13.0-21-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/anonymed06/Documents/Netgear-A6210-master/os/linux modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.13.0-21-generic »
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/assoc.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/auth.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/sync.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/sanity.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/connect.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/wpa.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../sta/sta.o
  CC [M]  /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:8:0,
                 from ./include/linux/cpumask.h:11,
                 from ./arch/x86/include/asm/cpumask.h:4,
                 from ./arch/x86/include/asm/msr.h:10,
                 from ./arch/x86/include/asm/processor.h:20,
                 from ./arch/x86/include/asm/cpufeature.h:4,
                 from ./arch/x86/include/asm/thread_info.h:52,
                 from ./include/linux/thread_info.h:37,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:80,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/anonymed06/Documents/Netgear-A6210-master/include/os/rt_linux.h:14,
                 from /home/anonymed06/Documents/Netgear-A6210-master/include/rtmp_os.h:30,
                 from /home/anonymed06/Documents/Netgear-A6210-master/include/rtmp_comm.h:64,
                 from /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:305:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
scripts/Makefile.build:302 : la recette pour la cible « /home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o » a échouée
make[2]: *** [/home/anonymed06/Documents/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o] Erreur 1
Makefile:1546 : la recette pour la cible « _module_/home/anonymed06/Documents/Netgear-A6210-master/os/linux » a échouée
make[1]: *** [_module_/home/anonymed06/Documents/Netgear-A6210-master/os/linux] Erreur 2
make[1] : on quitte le répertoire « /usr/src/linux-headers-4.13.0-21-generic »
Makefile:59 : la recette pour la cible « debug » a échouée
make: *** [debug] Erreur 2
`

TomLisankie, 2018-01-13:
I am getting the same problem.

dylan7581, 2018-01-13:
finally, i succeeded. you have to do "make clean" on your directory ("Netgear...") and you must remove your directory, then you must download this directory : https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.14 and you make directory then you make install your directory.
But now, i have an other problem : I still can not use it.

dylan7581, 2018-01-13:
Hi @jurobystricky 

I have problem, when i do "modprobe mt7662u_sta", it writes me : 
> modprobe: FATAL: Module mt7662u_sta not found in directory /lib/modules/4.13.0-25-generic

and in the file /etc/modules-load.d/modules.conf : the only lines are : 
```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
```
i have just make & make install Netgear-A6210-port-to-4.14 on my Documents files. And I do not know what to do, please help me.

surGeonGG, 2018-01-15:
> finally, i succeeded. you have to do "make clean" on your directory ("Netgear...") and you must remove your directory, then you must download this directory : https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.14 and you make directory then you make install your directory.

This worked for me except I used the port-to-4.13 branch since that's the kernel I have. Thanks, man!

dylan7581, 2018-01-15:
> I used the port-to-4.13 branch since that's the kernel I have.

Many thanks @surGeonGG,
My kernel is 4.13 and me i had download the 4.14! XD

dylan7581, 2018-01-15:
I just have a problem, when I try to connect the wifi, my screen gets stuck. :'-(

dylan7581, 2018-01-16:
@surGeonGG, did you do something else than "make" and "make install"?

surGeonGG, 2018-01-16:
@dylan7581 nothing except "git checkout port-to-4.13" and "make -B".

dylan7581, 2018-01-26:
@abdujaparov, what is your kernel (write the return of "uname -r").

abdujaparov, 2018-01-26:
uname -r
4.13.0-17-generic

dylan7581, 2018-01-26:
Okay,
"make clean" your directory (Netgear-A6210). Then 
`$ git clone https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13`
`$ cd Netgear-A6210-port-to-4.13/`
`$ make`
`$ sudo make install`

abdujaparov, 2018-01-27:
git clone https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13
Cloning into 'port-to-4.13'...
remote: Not Found
fatal: repository 'https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13/' not found

Git cloning failed but downloading zip from repository and following your instruction I succeded.
Thanks a lot!

TrishantRoy, 2018-01-27:
`trishant@EurusHolmes:~/Netgear-A6210-port-to-4.13$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/trishant/Netgear-A6210-port-to-4.13/os/linux/Makefile
make -C /lib/modules/4.13.0-31-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/trishant/Netgear-A6210-port-to-4.13/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-31-generic'
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/assoc.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/auth.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/sync.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/sanity.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/connect.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/wpa.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../sta/sta.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../os/linux/sta_ioctl.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../mgmt/mgmt_vht.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/vht.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/crypt_md5.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/crypt_sha2.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/crypt_hmac.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/crypt_aes.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/crypt_arc4.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/mlme.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_wep.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/action.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_data.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/rtmp_init.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/rtmp_init_inf.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_tkip.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_aes.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_sync.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/eeprom.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_sanity.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_info.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_cfg.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_wpa.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_radar.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/spectrum.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/rtmp_timer.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/rt_channel.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_profile.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_asic.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/scan.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_cmd.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/ps.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/sys_log.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/txpower.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../rate_ctrl/ra_ctrl.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../rate_ctrl/alg_legacy.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../rate_ctrl/alg_ags.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../chips/rtmp_chip.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../mgmt/mgmt_entrytb.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../tx_rx/wdev_tx.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../os/linux/rt_profile.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../phy/phy.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../phy/rf.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../phy/rlt_phy.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../mac/rtmp_mac.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../mac/ral_nmac.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../rate_ctrl/alg_grp.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/ba_action.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../mgmt/mgmt_ht.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_txbf.o
  CC [M]  /home/trishant/Netgear-A6210-port-to-4.13/os/linux/../../common/cmm_txbf_cal.o
[... cut ...]
cp /home/trishant/Netgear-A6210-port-to-4.13/conf/RT2870STA.dat /etc/Wireless/RT2870STA/.
install -d /lib/modules/4.13.0-31-generic/kernel/drivers/net/wireless/
install -m 644 -c mt7662u_sta.ko /lib/modules/4.13.0-31-generic/kernel/drivers/net/wireless/
/sbin/depmod -a 4.13.0-31-generic
make[1]: Leaving directory '/home/trishant/Netgear-A6210-port-to-4.13/os/linux'
`

@dylan7581  That worked for me also, I think.
But when I connect the Wifi USB adaptor, it isn't being displayed in the Wifi Connections.
@abdujaparov  Is yuor working?

Thanks in advance.

abdujaparov, 2018-01-27:
updated: I succeeded to connect but the connection is very very slow. In windows 10 with speedtest I get 180mbps in ubuntu less then 5mbps.
I updated ubuntu now but driver does not work anymore.
Can I install again?

dylan7581, 2018-01-27:
I think.

TrishantRoy, 2018-01-27:
@abdujaparov  Do tell yours is working?
Mine isn't detecting also.

Am I missing something?
I did this:
"make clean" your directory (Netgear-A6210). Then
$ git clone https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.13
$ cd Netgear-A6210-port-to-4.13/
$ make
$ sudo make install

tahnik, 2018-01-28:
@TrishantRoy I did the same and my adapter is not even recognized.

abdujaparov, 2018-01-28:
@TrishantRoy I installed again and the adapter is recognized but performances are not good around 13mbps, in windows I go over 150mbps in download

dylan7581, 2018-02-04:
I have resolve my problem, it's just my modem who is capricious.
And after make and sudo make install, it is necessary to execute sudo modprobe mt7662u_sta

sarsorus, 2019-05-22:
@dylan7581 Do you mean that the screen freeze issue was due to a faulty modem ?

dylan7581, 2019-05-22:
@sarsorus, if I remember correctly, the connection to my modem was the only freeze my screen (I have not tested on any other modem), so I think the solution (since I no longer use the netgear) was either to change my wifi dongle to the dongle: tnb 600mbps wifi dongle, or by upgrade my ubuntu to 18.04.

sarsorus, 2019-05-26:
Thanks. I couldn't get the driver to work on my Ubuntu 18.04, but with Ubuntu 19 (kernel 5.0) and it works natively.

closed 2021-11-21 by dylan7581