jurobystricky/Netgear-A6210 #99: USB Dongle not recognized

Freakazoidx86, 2018-01-25:
Hi, I don't have that much experience coding, I do have coded in other languages but nothing strong on C like this; anyhow, I used the fork by Thog, which is for Kernel 4.14; Everything compiles fine, I do have a warning about something being Not Null, other than that everything seems ok; I make install without errors either, modprobe the module, but when I insert the dongle, this is shown on my logs

journal:

> Jan 25 13:55:02 archBox kernel: usb 1-5: new high-speed USB device number 2 using ehci-pci
> Jan 25 13:55:02 archBox mtp-probe[1135]: checking bus 1, device 2: "/sys/devices/pci0000:00/0000:00:12.2/usb1/1-5"
> Jan 25 13:55:02 archBox mtp-probe[1135]: bus: 1, device: 2 was not an MTP device
> Jan 25 13:55:02 archBox colord[662]: /usr/lib/colord-sane: error while loading shared libraries: libsane.so.1: cannot open shared object file: No such file or directory
> Jan 25 13:55:02 archBox upowerd[550]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:12.2/usb1/1-5

dmesg:

> [   50.328774] rtusb init rt2870 --->
> [   50.328812] usbcore: registered new interface driver rt2870
> [   55.126663] usb 2-5: new high-speed USB device number 5 using ehci-pci
> [  116.093708] usb 2-5: USB disconnect, device number 5
> [  125.609989] usb 2-5: new high-speed USB device number 6 using ehci-pci
> [  173.664756] usb 2-5: USB disconnect, device number 6
> [  181.906631] usb 1-5: new high-speed USB device number 2 using ehci-pci
> [  469.644142] usb 1-5: USB disconnect, device number 2
> [  474.569939] usb 1-5: new high-speed USB device number 3 using ehci-pci

And its just not recognized, i get no wlan0 device

My OS is Archlinux 64Bits with Kernel 4.14
If any information is needed just let me know, I have to boot Windows in order to get Internet from this PC.
Many thanks in advance
PS: English is not my native language, I apologize if grammatical errors or any sentence makes no sense :/

dylan7581, 2018-01-26:
Hi @Freakazoidx86, did you make and make install this directory : https://github.com/genodeftest/Netgear-A6210/tree/port-to-4.14, Instead of jurobystricky?

Freakazoidx86, 2018-01-26:
I tried that dir a couple of days before but I get some warnings about vfs_read and vfs_writte, that even as simple warnings, when I make install and plug in the dongle, I get some errors on dmesg, I don't remember exactly right now but they were related to the warnings, it's 2am in the morning in here lol, tomorrow I can reproduce those errors and come back with the logs.

Many thanks for replying

Freakazoidx86, 2018-01-26:
Hi @dylan7581 Here are the logs after using  **genodeftest** directory:
I append the _genode_ string in the folder name, just to differ from the other directory I previously used(Thog)

make:
[make log.txt](https://github.com/jurobystricky/Netgear-A6210/files/1669377/make.log.txt)

clean sudo make install:
`[reoba@archBox Netgear-A6210-4.14-genode]$ sudo make install
[sudo] password for reoba: 
make -C /home/reoba/Netgear-A6210-4.14-genode/os/linux -f Makefile.6 install
make[1]: Entering directory '/home/reoba/Netgear-A6210-4.14-genode/os/linux'
rm -rf /etc/Wireless/RT2870STA
mkdir -pv /etc/Wireless/RT2870STA
mkdir: created directory '/etc/Wireless/RT2870STA'
cp /home/reoba/Netgear-A6210-4.14-genode/conf/RT2870STA.dat /etc/Wireless/RT2870STA/.
install -d /lib/modules/4.14.13-1-ARCH/kernel/drivers/net/wireless/
install -m 644 -c mt7662u_sta.ko /lib/modules/4.14.13-1-ARCH/kernel/drivers/net/wireless/
/sbin/depmod -a 4.14.13-1-ARCH
make[1]: Leaving directory '/home/reoba/Netgear-A6210-4.14-genode/os/linux'`

And then, I can't load the module mt7662u_sta:
`[reoba@archBox Netgear-A6210-4.14-genode]$ sudo modprobe mt7662u_sta
modprobe: ERROR: could not insert 'mt7662u_sta': Unknown symbol in module, or unknown parameter (see dmesg)`

dmesg:
`[ 1192.020936] mt7662u_sta: Unknown symbol vfs_write (err 0)
[ 1192.021186] mt7662u_sta: Unknown symbol vfs_read (err 0)`

I noticed that even without the driver loaded, if I plug-in the dongle, I get the same journal lines that I post in first post, does that means driver isnt really loaded in both cases?
I feel somehow pretty lost here lol btw, I downloaded the genodeftest directory from another PC using this command, is it right?

`git clone -b port-to-4.14 --single-branch https://github.com/genodeftest/Netgear-A6210.git`

Then i just copied the folder with a usb.

dylan7581, 2018-01-27:
Hi @Freakazoidx86 ,
Sorry I wanted to help you at best, but I can make nothing furthermore, I even I did not solve my problem.

Freakazoidx86, 2018-01-27:
@dylan7581 Oh It's ok no worries, the info is there for whoever might know what my problem is. 

Many thanks for replying man, Greetings!

dylan7581, 2018-01-27:
;-) Greetings, good luck!

Freakazoidx86, 2018-02-06:
I think I know why it doesnt work...  I thought my dongle had the same chip as the Mediatek used in the AC6210, but it doesn't, my A6200 has the Vendor and Device IDs 0846:9050 wich uses the Broadcom BCM43526 with no support at all for linux.... So, it sucks but its my bad for buying something without previous investigation.

Freakazoidx86, 2018-02-24:
I will just comment that I found a some sort of fix.... I use a VM with either Windows XP or Windows 7, just to bypass the Internet from the VM to Arch, I thought maybe this could be useful to someone else.