jurobystricky/Netgear-A6210 #100: Nee to unplug and replug every time upload large files

yeze16159, 2018-01-29:
Hi, everyone.
I recently get some rare issues regarding to my netgear a6210 wireless adapter.
Before i started, there is some back ground need to explained:
The operating system is ubuntu 17.10 dual booth with windows 10. 
The kernel version is 4.13.0. The CPU is i7-8700k and graphical card is NVIDIA 1080 ti (In case that this is actually causing problem).
Here we go:
Once i install  Netgear-A6210-port-to-4.13.zip driver in my OS, and run 
`sudo service network-manager restart`
I was able to connect to the internet.
And here is the current state of dmesg, and status of nm
[dmesg.log](https://github.com/jurobystricky/Netgear-A6210/files/1672193/dmesg.log)
[nm.log](https://github.com/jurobystricky/Netgear-A6210/files/1672195/nm.log)
However, after uploading large files for several second, the wifi is disconnected and if i run
`nm-applet`
It wiil show that device not ready.
Also, in nm status, it show wpa_supplicant couldn't grab the interface Here is the current state of dmesg and nm log
[dmesg1.log](https://github.com/jurobystricky/Netgear-A6210/files/1672204/dmesg1.log)
[nm1.log](https://github.com/jurobystricky/Netgear-A6210/files/1672206/nm1.log)

At this point I tried everything i can to make it functional, but failed at the end. (even with reboot)
The only way i can solve this issue is unplug the adapter and replug it again. 
After that, i restart nm and it back to work again! 
[nm2.log](https://github.com/jurobystricky/Netgear-A6210/files/1672209/nm2.log)

So can anyone help me with this annoying bug?

zo7, 2018-03-01:
I am experiencing a similar problem, except in my case I am uploading a large Docker image.

Since you're using Ubuntu 17.10 / Linux 4.13.0, are you using the 4.13 port in #83? I'm also using 17, but I wonder if this issue is specific to that port or 4.13.

yeze16159, 2018-03-01:
Hi, i don't remember which port is in use. 
After days of trying, I gave up on this wifi card and bought a new one and everything works charm...

zo7, 2018-03-01:
Hah, makes sense. Which wifi card did you end up using?

I got this Netgear card because someone recommended it to me for Linux, but so far it's just been a headache.