jurobystricky/Netgear-A6210 #102: ‘__read_overflow2’ problem with kernal 14.13

maveryKearney, 2018-02-26:
I've looked through recent related issues. It appears the following should work and did with prior kernals,   Just not with this . Make log is included below. 

`~/Downloads/Netgear-A6210-master $ uname -r
4.13.0-36-generic

~/Downloads/Netgear-A6210-master $ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/Makefile
make -C /lib/modules/4.13.0-36-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/arch_admin/Downloads/Netgear-A6210-master/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.13.0-36-generic'
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/assoc.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/auth.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/sync.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/sanity.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/connect.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/wpa.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../sta/sta.o
  CC [M]  /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:8:0,
                 from ./include/linux/cpumask.h:11,
                 from ./arch/x86/include/asm/cpumask.h:4,
                 from ./arch/x86/include/asm/msr.h:10,
                 from ./arch/x86/include/asm/processor.h:20,
                 from ./arch/x86/include/asm/cpufeature.h:4,
                 from ./arch/x86/include/asm/thread_info.h:63,
                 from ./include/linux/thread_info.h:37,
                 from ./arch/x86/include/asm/preempt.h:6,
                 from ./include/linux/preempt.h:80,
                 from ./include/linux/spinlock.h:50,
                 from ./include/linux/seqlock.h:35,
                 from ./include/linux/time.h:5,
                 from ./include/linux/stat.h:18,
                 from ./include/linux/module.h:10,
                 from /home/arch_admin/Downloads/Netgear-A6210-master/include/os/rt_linux.h:14,
                 from /home/arch_admin/Downloads/Netgear-A6210-master/include/rtmp_os.h:30,
                 from /home/arch_admin/Downloads/Netgear-A6210-master/include/rtmp_comm.h:64,
                 from /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:305:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^
scripts/Makefile.build:308: recipe for target '/home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/home/arch_admin/Downloads/Netgear-A6210-master/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1550: recipe for target '_module_/home/arch_admin/Downloads/Netgear-A6210-master/os/linux' failed
make[1]: *** [_module_/home/arch_admin/Downloads/Netgear-A6210-master/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.13.0-36-generic'
Makefile:59: recipe for target 'debug' failed

`

dmattox10, 2018-02-28:
I get this as well.

zo7, 2018-03-01:
There is a PR at #83 that seems to work, last merge to this repo was in April 2017 so I'm not sure this is being maintained anymore.

inkplay, 2018-03-27:
Same issue, worst part is I can't get internet on my desktop until I can get this driver working and I started using ubuntu...

maveryKearney, 2018-03-28:
Fortunately for now I can just hard wire my machine to the router. I have three machines though where the final application is dependent on wifi, and I had purchased  Netgear A6210s .  My mistake. Netgear does not claim to support linux. It is nice that folks do hack drivers that work. IMO It is better to use hardware from a company that has committed support for linux.