jurobystricky/Netgear-A6210 #104: on CentOS linux 7 error 2 and error 1 for make

Cwoodyard, 2018-04-07:
i am currently running centOS 7 and having the following make errors:

[admin@localhost Desktop]$ git clone https://github.com/jurobystricky/Netgear-A6210
Cloning into 'Netgear-A6210'...
remote: Counting objects: 2436, done.
remote: Total 2436 (delta 0), reused 0 (delta 0), pack-reused 2436
Receiving objects: 100% (2436/2436), 3.71 MiB | 1002.00 KiB/s, done.
Resolving deltas: 100% (1512/1512), done.
[admin@localhost Desktop]$ cd Netgear-A6210
[admin@localhost Netgear-A6210]$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/admin/Desktop/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/3.10.0-693.el7.x86_64/build DBGFLAGS=-DDBG SUBDIRS=/home/admin/Desktop/Netgear-A6210/os/linux modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-693.el7.x86_64'
  CC [M]  /home/admin/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o
In file included from /home/admin/Desktop/Netgear-A6210/include/os/rt_linux.h:77:0,
                 from /home/admin/Desktop/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/admin/Desktop/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/admin/Desktop/Netgear-A6210/include/rt_config.h:34,
                 from /home/admin/Desktop/Netgear-A6210/os/linux/../../sta/assoc.c:28:
/home/admin/Desktop/Netgear-A6210/include/cfg80211.h:45:49: error: ‘IEEE80211_NUM_BANDS’ undeclared here (not in a function)
  struct ieee80211_supported_band Cfg80211_bands[IEEE80211_NUM_BANDS];
                                                 ^
make[2]: *** [/home/admin/Desktop/Netgear-A6210/os/linux/../../sta/assoc.o] Error 1
make[1]: *** [_module_/home/admin/Desktop/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-693.el7.x86_64'
make: *** [debug] Error 2

OldJohn86, 2019-03-23:
I met the same issue.

agingo, 2019-09-07:
Same here. Does anyone have a solution yet?