jurobystricky/Netgear-A6210 #105: Build error on 4.15.14-200.fc26.x86_64

dsime42, 2018-04-08:
Using code from 'Netgear-A6210-port-to-4.14'
It seems related to 
**https://github.com/zfsonlinux/spl/issues/670**
and a fix of sorts seems to be here
**https://github.com/zfsonlinux/spl/commit/71513af020c0522560dc86668db46b49068787b3**

....../rt_linux.c: In function ‘__RTMP_OS_Init_Timer’:
....../rt_linux.c:107:3: error: implicit declaration of function ‘init_timer’; did you mean ‘init_timers’? 
[-Werror=implicit-function-declaration]
   init_timer(pTimer);
   ^~~~~~~~~~
   init_timers
....../rt_linux.c:108:9: error: ‘struct timer_list’ has no member named ‘data’
   pTimer->data = (unsigned long)data;
         ^~
....../rt_linux.c:109:20: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
   pTimer->function = function;
                    ^
cc1: some warnings being treated as errors

dsime42, 2018-04-08:
Systemtap reported the problem in December so I am probably late with this report
**https://sourceware.org/git/gitweb.cgi?p=systemtap.git;a=commitdiff;h=fbb26e17a**
I believe this is related to upstream kernel git commit 919b250f8570618e84af544c3e18dad5210eb9b6

They fixed it here
**https://sourceware.org/git/gitweb.cgi?p=systemtap.git;a=commitdiff;h=fbb26e17a**

Proximyst, 2018-04-09:
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15
It is in my experience glitchy as hell, though.. Might be worth looking into, but I haven't really bothered.

dsime42, 2018-04-09:
ThanksIt compiled and loadedWhat issues did you have that seem glitchy

   
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

wheelcomplex, 2018-04-09:
Thanks for your code.

FYI:

test https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15 with EP-AC1602: https://www.newegg.ca/Product/Product.aspx?item=9SIABSJ4V82410

set MODULE = mt7612u in Makefile

Ubuntu 18.04
4.15.0-13-generic #14-Ubuntu SMP Sat Mar 17 13:44:27 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

not SSID showed in network manager

dmesg:
<pre>
[ 1363.912838] rtusb init rt2870 --->
[ 1363.912877] usbcore: registered new interface driver rt2870
[ 1365.472625] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1365.473365] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1369.503662] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1372.532342] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1372.532600] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1376.567922] r8152 4-1.4:1.0 enx00e04c063287: Promiscuous mode enabled
[ 1377.583412] usb 4-2: new SuperSpeed USB device number 4 using xhci_hcd
[ 1377.604937] usb 4-2: New USB device found, idVendor=0e8d, idProduct=7612
[ 1377.604941] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1377.604943] usb 4-2: Product: 802.11ac WLAN
[ 1377.604945] usb 4-2: Manufacturer: MediaTek Inc.
[ 1377.604947] usb 4-2: SerialNumber: 000000000
[ 1377.606937] 
               === pAd = 000000005fa0a63a, size = 1241112 ===

[ 1377.606939] driver version: 3.0.0.1 (Apr 10 2018 09:31:32) .
[ 1377.607610] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
[ 1377.614476] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
[ 1377.614478] RtmpEepromGetDefault::e2p_dafault=1
[ 1377.614653] NVM is EFUSE mode
[ 1377.614654] Endpoint(8) is for In-band Command
[ 1377.614656] Endpoint(4) is for WMM0 AC0
[ 1377.614657] Endpoint(5) is for WMM0 AC1
[ 1377.614658] Endpoint(6) is for WMM0 AC2
[ 1377.614659] Endpoint(7) is for WMM0 AC3
[ 1377.614660] Endpoint(9) is for WMM1 AC0
[ 1377.614661] Endpoint(84) is for Data-In
[ 1377.614662] Endpoint(85) is for Command Rsp
[ 1377.614776] 80211> CurTxPower = 20 dBm
[ 1377.614778] ====> Radar Channel 52
[ 1377.614779] ====> Radar Channel 54
[ 1377.614780] ====> Radar Channel 56
[ 1377.614781] ====> Radar Channel 60
[ 1377.614781] ====> Radar Channel 62
[ 1377.614782] ====> Radar Channel 64
[ 1377.614783] ====> Radar Channel 100
[ 1377.614784] ====> Radar Channel 104
[ 1377.614785] 80211> TxStream = 0
[ 1377.634641] rtusb init rt2870 --->
[ 1377.634644] debugfs_create_dir failed
[ 1377.634651] Error: Driver 'rt2870' is already registered, aborting...
[ 1377.638991] show_signal_msg: 34 callbacks suppressed
[ 1377.638994] ModemManager[4084]: segfault at 0 ip 0000561907426659 sp 00007ffde73c26e0 error 4 in ModemManager[5619073eb000+123000]
</pre>

ifconfig -a
<pre>
wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 00:00:00:00:00:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
</pre>

wheelcomplex, 2018-04-10:
if keep wifi dongle plugin  when system booting up, the adapter works for a while and stop works with some debug message:

[dmesg.log](https://github.com/jurobystricky/Netgear-A6210/files/1892431/dmesg.log)

dsime42, 2018-04-10:
Eh, Nevermind I think I know what you mean by glitchy
I had to replug the USB to get it back after it just sort of went away

      From: Donald Sime <dsime42@yahoo.com>
 To: jurobystricky/Netgear-A6210 <reply@reply.github.com> 
 Sent: Monday, April 9, 2018 10:28 AM
 Subject: Re: [jurobystricky/Netgear-A6210] Build error on 4.15.14-200.fc26.x86_64 (#105)
   
ThanksIt compiled and loadedWhat issues did you have that seem glitchy

  
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

dsime42, 2018-04-13:
I may have found a "fix" to "glitchy"
I looked at the errors and they seemed to be similar to some problems from'15 and '16 that were caused by NetworkManager
I checked my configuration and found that "something" has re-enabled my IPv6
I re-disabled IPv6 and the interface has been running OK for a while. 
I hope this fixes it until something re-enabled my IPv6 again for no reason.

dsime42, 2018-07-06:
Build error on 4.15.14 using code from 
Netgear-A6210-port-to-4.14
Was resolved by updating/upgrading to 
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

closed 2018-07-06 by dsime42