jurobystricky/Netgear-A6210 #107: Build error on 4.15.15 on Arch

andrejpodzimek, 2018-04-18:
```
$ LC_ALL=C make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/andrej/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.15-1-user-regd/build DBGFLAGS=-DDBG SUBDIRS=/home/andrej/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.15.15-1-user-regd/build'
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /home/andrej/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9:0,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/andrej/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/andrej/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/andrej/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/andrej/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/andrej/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:340:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:325: /home/andrej/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Chyba 1
make[1]: *** [Makefile:1524: _module_/home/andrej/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.15.15-1-user-regd/build'
make: *** [Makefile:64: debug] Error 2
```

dsime42, 2018-04-20:
This one works
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

But not well. :-)

daylanc, 2018-10-02:
@dsime42 Once installed, what else do I have to do? it seems like `sudo make install` worked properly, but when I disconnect the Ethernet cable, I don't have wifi. Am I missing a step after?

Nagefire, 2018-10-14:
https://wiki.archlinux.org/index.php/Network_configuration