jurobystricky/Netgear-A6210 #108: Success with kernel 14.15 / Ubuntu 18.04 (b2)

dannynicolas, 2018-04-24:
I was able to get this working via the 14.15 branch (https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15). I'm on the latest beta release of 18.04 Ubuntu (64 bit / 14.15.0-19-generic kernel), with a fresh install.

closed 2018-04-24 by dannynicolas

ajsingh007, 2018-05-07:
I just got ubuntu 18.04 LTS and was able to successfully make and make install using the above link. However, I still get the **No Wi-Fi Adapter Found** message in Settings after restarting and running `sudo service network-manager restart`

Any insight or advice? Thanks!

dannynicolas, 2018-05-07:
I would unplug the USB device, then plug it in and see if it recognizes it. if it doesn't see the device still, try restarting again.

ajsingh007, 2018-05-07:
Appreciate the rapid response! That literally worked right now, perhaps because I also disabled secure boot? Anyway, thank you for posting the 4.15 -- it is what I needed!

lahans, 2018-07-03:
nice work. For 17.x I had to tweak the code myself to get it working:)

Has anyone experienced after the computer is idling, that network stops working? I don't get much in the log files other than wlan0: link not ready. Then I restart my machine in order to get network again:(

I have tried take out the adapter and put it back in without luck

MiscellaneousMemphian, 2018-08-18:
Yes I can not get any wireless network to appear I am using Ubuntu 18.04 LTS.

krdeepak, 2019-08-05:
Did someone get it working for latest ubuntu 18.04.2 which has kernel 4.18.0-15-generic ?