jurobystricky/Netgear-A6210 #112: This is good on 4.17.x https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

Joseph-Smith, 2018-07-05:
Just in case people are looking for something that functions on 4.16+     
https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15

rickytickytabby, 2018-07-28:
Awesome! Thank you. Still having issues however, when going into airmon-ng the tab that shows "drivers" is only showing "??????". My adapter works fine for a while, allowing me to connect to the internet, but after a little while my terminal will become unresponsive if I issue any commands having to do with the network. airmon-ng, iwconfig, ifconfig..... if I issue any of these commands, nothing happens and I have to restart my system if I want the damage to be un-done. The issue of the un-responsive terminal will only happen after I plug in my adapter and about 5 minutes has gone by.

Any guesses on what is causing this? Is there more information I should include?

xgladius, 2018-08-01:
worked for me, connected using my phone and a tethered connection.
download zip from https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15
unpack
cd
make
sudo apt install
restart

Nagefire, 2018-10-14:
Note that it's also backwards compatible with kernels preceeding 4.16 because the function that was breaking compatibility for 4.16+ was the a timer changing name, and the commit that fixed it used boilerplate code. The memcpy bug also got fixed about 4 commits back, which would fix #120 #111 #116 #107 and several other issues, I imagine