jurobystricky/Netgear-A6210 #113: cant figure this out please help new to linux

jeffand123, 2018-07-20:
*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.16.0-kali2-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.16.0-kali2-amd64'
  CC [M]  /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/bitmap.h:9:0,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/cpumask.h:12,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/cpumask.h:5,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/msr.h:11,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/processor.h:21,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/cpufeature.h:5,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/thread_info.h:53,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/thread_info.h:38,
                 from /usr/src/linux-headers-4.16.0-kali2-common/arch/x86/include/asm/preempt.h:7,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/preempt.h:81,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/spinlock.h:51,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/seqlock.h:36,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/time.h:6,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/stat.h:19,
                 from /usr/src/linux-headers-4.16.0-kali2-common/include/linux/module.h:10,
                 from /root/Netgear-A6210/include/os/rt_linux.h:14,
                 from /root/Netgear-A6210/include/rtmp_os.h:30,
                 from /root/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
/usr/src/linux-headers-4.16.0-kali2-common/include/linux/string.h:341:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.16.0-kali2-common/scripts/Makefile.build:330: /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.16.0-kali2-common/Makefile:1578: _module_/root/Netgear-A6210/os/linux] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.16.0-kali2-amd64'
make: *** [Makefile:64: debug] Error 2

PengYicong, 2018-07-25:
It's the same issue opened before, I also get this read overflow problem when I tried to make the driver.

- I don't find the solution so far, maybe it's because the kernel version is different.

- I also tried the MTK MT7612U driver from mediatek website, it gave me a different error report when I tried to make the file. Something about IEEE band specification.

t0ny-fr33man, 2018-08-20:
I have been getting the same error for months.  Here's the latest build attempt on an ubuntu 64 system:

# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /usr/src/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.0-32-generic/build DBGFLAGS=-DDBG SUBDIRS=/usr/src/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-32-generic'
  CC [M]  /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9:0,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /usr/src/Netgear-A6210/include/os/rt_linux.h:14,
                 from /usr/src/Netgear-A6210/include/rtmp_os.h:30,
                 from /usr/src/Netgear-A6210/include/rtmp_comm.h:64,
                 from /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:340:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^
scripts/Makefile.build:332: recipe for target '/usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/usr/src/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1552: recipe for target '_module_/usr/src/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/usr/src/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-32-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

PengYicong, 2018-08-21:
Update: I got mine working pretty nicely now. 
Two things you might need to notice though:1. The kernel version of your Linux (which can be identified by using the uname -a command in the terminal. ) and download the corresponding port of this driver. For example: I’m running an 18.04 Ubuntu on my computer with a kernel version of 4.15 then I should download the port to 415 branch to compile. 2. You need to turn off secure boot in your bios for this driver to take effect.  Reply if needed. Good luck man.

kermesvermilio, 2019-01-22:
> Update: I got mine working pretty nicely now.
> Two things you might need to notice though:1. The kernel version of your Linux (which can be identified by using the uname -a command in the terminal. ) [...]

Where did you get the port to 415 branch? All I can see is a master branch, and that gives me the read overflow error ;-/

BLVCK971, 2019-11-02:
The update for kernel 4.15 and up is at https://github.com/kaduke/Netgear-A6210 162
```
rm -rf ~/Netgear-A6210
git clone https://github.com/kaduke/Netgear-A6210
cd Netgear-A6210
make
sudo make install
```