jurobystricky/Netgear-A6210 #115: failure on Kernel-4.15.0-29-generic Ubuntu 16.04

jack338c, 2018-07-29:
jack@jack-R478-R429:~/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/jack/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.0-29-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/jack/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-29-generic'
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../mgmt/mgmt_vht.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/vht.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/crypt_md5.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/crypt_sha2.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/crypt_hmac.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/crypt_aes.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/crypt_arc4.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/mlme.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_wep.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/action.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_data.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/rtmp_init.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/rtmp_init_inf.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_tkip.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_aes.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_sync.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/eeprom.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_sanity.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_info.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_cfg.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_wpa.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_radar.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/spectrum.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/rtmp_timer.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/rt_channel.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_profile.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_asic.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/scan.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_cmd.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/ps.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/sys_log.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/txpower.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../rate_ctrl/ra_ctrl.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../rate_ctrl/alg_legacy.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../rate_ctrl/alg_ags.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../chips/rtmp_chip.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../mgmt/mgmt_entrytb.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../tx_rx/wdev_tx.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../os/linux/rt_profile.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../phy/phy.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../phy/rf.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../phy/rlt_phy.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../mac/rtmp_mac.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../mac/ral_nmac.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../rate_ctrl/alg_grp.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/ba_action.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../mgmt/mgmt_ht.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_txbf.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/cmm_txbf_cal.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../common/rt_os_util.o
  CC [M]  /home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.o
/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.c: In function ‘__RTMP_OS_Init_Timer’:
/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.c:107:3: error: implicit declaration of function ‘init_timer’ [-Werror=implicit-function-declaration]
   init_timer(pTimer);
   ^
/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.c:108:9: error: ‘struct timer_list’ has no member named ‘data’
   pTimer->data = (unsigned long)data;
         ^
/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.c:109:20: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
   pTimer->function = function;
                    ^
cc1: some warnings being treated as errors
scripts/Makefile.build:332: recipe for target '/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.o' failed
make[2]: *** [/home/jack/Netgear-A6210/os/linux/../../os/linux/rt_linux.o] Error 1
Makefile:1552: recipe for target '_module_/home/jack/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/jack/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-29-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2
jack@jack-R478-R429:~/Netgear-A6210$

kaduke, 2018-08-12:
Hi Jack, this was caused by a breaking change in version 4.15 of the kernel. My fork currently has been tested on up to 4.17, feel free to try it out.

https://github.com/kaduke/Netgear-A6210

daylanc, 2018-10-02:
@kaduke It seems like I installed the driver correctly.. My kernel is 4.15.0.36. I can successfully run `sudo make install` but when I disconnect my ethernet cable, nothing happens. I tried rebooting and still no wifi. Am I missing a step after installing? I am using Ubuntu 16.04 with the previously mentioned kernel. 
I am using an X299 motherboard and my secure boot is "disabled" in the "Windows OS Configuration" menu. I don't know if this means that secure boot is disabled in ubuntu also, but this is the only place I could find "secure boot" in the bios. 
I hope you can help.
Thanks!

jack338c, 2018-10-18:
https://askubuntu.com/questions/1033772/cant-get-my-netgear-a6210-usb-wifi-adapter-to-work-on-my-linux-ubuntu

this help me

closed 2018-10-18 by jack338c