jurobystricky/Netgear-A6210 #116: memcpy bug on 4.17.0-kali1-amd64

luiginesbitt, 2018-08-07:
Tried the fix suggested by genodeftest:
`	memcpy(extra, addr, i*sizeof(struct sockaddr));

instead of

	memcpy(extra, addr, i*sizeof(addr[0]));
`

`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.17.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.17.0-kali1-amd64'
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /root/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/bitmap.h:9,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/cpumask.h:12,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/cpumask.h:5,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/msr.h:11,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/processor.h:21,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/cpufeature.h:5,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/thread_info.h:53,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/thread_info.h:38,
                 from /usr/src/linux-headers-4.17.0-kali1-common/arch/x86/include/asm/preempt.h:7,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/preempt.h:81,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/spinlock.h:51,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/seqlock.h:36,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/time.h:6,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/stat.h:19,
                 from /usr/src/linux-headers-4.17.0-kali1-common/include/linux/module.h:10,
                 from /root/Netgear-A6210/include/os/rt_linux.h:14,
                 from /root/Netgear-A6210/include/rtmp_os.h:30,
                 from /root/Netgear-A6210/include/rtmp_comm.h:64,
                 from /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
/usr/src/linux-headers-4.17.0-kali1-common/include/linux/string.h:341:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-4.17.0-kali1-common/scripts/Makefile.build:318: /root/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[3]: *** [/usr/src/linux-headers-4.17.0-kali1-common/Makefile:1585: _module_/root/Netgear-A6210/os/linux] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:8: all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.17.0-kali1-amd64'
make: *** [Makefile:64: debug] Error 2`

luiginesbitt, 2018-08-20:
Still a no go on that fork kaduke, I can see my netgear in the network mangager but when I go to connect to a wireless ap it freezes the network manager. Tried reboot and unplug/plug back in also. Here are my logs:

Kern_log

`usb 4-1.1: New USB device found, idVendor=0846, idProduct=9053, bcdDevice= 1.00
Aug 14 20:02:54 Kryptonian kernel: [  114.570726] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Aug 14 20:02:54 Kryptonian kernel: [  114.570731] usb 4-1.1: Product: A6210
Aug 14 20:02:54 Kryptonian kernel: [  114.570735] usb 4-1.1: Manufacturer: NETGEAR
Aug 14 20:02:54 Kryptonian kernel: [  114.570739] usb 4-1.1: SerialNumber: 100
Aug 14 20:02:56 Kryptonian kernel: [  116.672525] mt7662u_sta: loading out-of-tree module taints kernel.
Aug 14 20:02:56 Kryptonian kernel: [  116.685651] rtusb init rt2870 --->
Aug 14 20:02:56 Kryptonian kernel: [  116.685832] 
Aug 14 20:02:56 Kryptonian kernel: [  116.685832] === pAd = 0000000083aecabb, size = 1241112 ===
Aug 14 20:02:56 Kryptonian kernel: [  116.685832] 
Aug 14 20:02:56 Kryptonian kernel: [  116.685833] driver version: 3.0.0.1 (Aug 14 2018 19:52:26) .
Aug 14 20:02:56 Kryptonian kernel: [  116.686434] ==>rlt_wlan_chip_onoff(): OnOff:1, Reset= 1, pAd->WlanFunCtrl:0x0, Reg-WlanFunCtrl=0x20a
Aug 14 20:02:56 Kryptonian kernel: [  116.698069] RtmpChipOpsEepromHook::e2p_type=0, inf_Type=2
Aug 14 20:02:56 Kryptonian kernel: [  116.698072] RtmpEepromGetDefault::e2p_dafault=1
Aug 14 20:02:56 Kryptonian kernel: [  116.698562] NVM is EFUSE mode
Aug 14 20:02:56 Kryptonian kernel: [  116.698565] Endpoint(8) is for In-band Command
Aug 14 20:02:56 Kryptonian kernel: [  116.698567] Endpoint(4) is for WMM0 AC0
Aug 14 20:02:56 Kryptonian kernel: [  116.698569] Endpoint(5) is for WMM0 AC1
Aug 14 20:02:56 Kryptonian kernel: [  116.698571] Endpoint(6) is for WMM0 AC2
Aug 14 20:02:56 Kryptonian kernel: [  116.698573] Endpoint(7) is for WMM0 AC3
Aug 14 20:02:56 Kryptonian kernel: [  116.698575] Endpoint(9) is for WMM1 AC0
Aug 14 20:02:56 Kryptonian kernel: [  116.698576] Endpoint(84) is for Data-In
Aug 14 20:02:56 Kryptonian kernel: [  116.698578] Endpoint(85) is for Command Rsp
Aug 14 20:02:56 Kryptonian kernel: [  116.698827] 80211> CurTxPower = 20 dBm
Aug 14 20:02:56 Kryptonian kernel: [  116.698831] ====> Radar Channel 52
Aug 14 20:02:56 Kryptonian kernel: [  116.698833] ====> Radar Channel 54
Aug 14 20:02:56 Kryptonian kernel: [  116.698834] ====> Radar Channel 56
Aug 14 20:02:56 Kryptonian kernel: [  116.698836] ====> Radar Channel 60
Aug 14 20:02:56 Kryptonian kernel: [  116.698837] ====> Radar Channel 62
Aug 14 20:02:56 Kryptonian kernel: [  116.698839] ====> Radar Channel 64
Aug 14 20:02:56 Kryptonian kernel: [  116.698841] ====> Radar Channel 100
Aug 14 20:02:56 Kryptonian kernel: [  116.698842] ====> Radar Channel 104
Aug 14 20:02:56 Kryptonian kernel: [  116.698845] 80211> TxStream = 0
Aug 14 20:02:56 Kryptonian kernel: [  116.699338] usbcore: registered new interface driver rt2870
Aug 14 20:02:56 Kryptonian kernel: [  116.718439] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Aug 14 20:02:56 Kryptonian kernel: [  116.719184] -->RTUSBVendorReset
Aug 14 20:02:56 Kryptonian kernel: [  116.739684] <--RTUSBVendorReset
Aug 14 20:02:56 Kryptonian kernel: [  116.740926] .
Aug 14 20:02:56 Kryptonian kernel: [  116.746802] .
Aug 14 20:02:56 Kryptonian kernel: [  116.752677] .
Aug 14 20:02:56 Kryptonian kernel: [  116.758552] .
Aug 14 20:02:56 Kryptonian kernel: [  116.764552] .
Aug 14 20:02:56 Kryptonian kernel: [  116.770427] .
Aug 14 20:02:56 Kryptonian kernel: [  116.776302] .
Aug 14 20:02:56 Kryptonian kernel: [  116.782177] .
Aug 14 20:02:56 Kryptonian kernel: [  116.788053] .
Aug 14 20:02:56 Kryptonian kernel: [  116.793928] .
Aug 14 20:02:56 Kryptonian kernel: [  116.799808] .
Aug 14 20:02:56 Kryptonian kernel: [  116.809994] andes_usb_chk_crc
Aug 14 20:02:56 Kryptonian kernel: [  116.833825] andes_usb_reset_wmt
Aug 14 20:02:56 Kryptonian kernel: [  116.854323] -->RTUSBVendorReset
Aug 14 20:02:56 Kryptonian kernel: [  116.874291] <--RTUSBVendorReset
Aug 14 20:02:56 Kryptonian kernel: [  116.874556] fw version:0.0.00 
Aug 14 20:02:56 Kryptonian kernel: [  116.874558] build:1
Aug 14 20:02:56 Kryptonian kernel: [  116.874558] build time:
Aug 14 20:02:56 Kryptonian kernel: [  116.874559] 2
Aug 14 20:02:56 Kryptonian kernel: [  116.874560] 0
Aug 14 20:02:56 Kryptonian kernel: [  116.874560] 1
Aug 14 20:02:56 Kryptonian kernel: [  116.874561] 4
Aug 14 20:02:56 Kryptonian kernel: [  116.874561] 0
Aug 14 20:02:56 Kryptonian kernel: [  116.874561] 6
Aug 14 20:02:56 Kryptonian kernel: [  116.874562] 2
Aug 14 20:02:56 Kryptonian kernel: [  116.874562] 4
Aug 14 20:02:56 Kryptonian kernel: [  116.874563] 1
Aug 14 20:02:56 Kryptonian kernel: [  116.874563] 8
Aug 14 20:02:56 Kryptonian kernel: [  116.874563] 3
[... 669 lines cut ...]
Aug 14 20:02:58 Kryptonian kernel: [  117.960422] TX1 power compensation = 0x38
Aug 14 20:02:58 Kryptonian kernel: [  117.968743] PeerBeaconAtScanAction P2P_SCANNING: DIRECT-64-HP OfficeJet Pro 8710 [1]
Aug 14 20:02:58 Kryptonian kernel: [  117.973099] PeerBeaconAtScanAction P2P_SCANNING: DIRECT-64-HP OfficeJet Pro 8710 [1]
Aug 14 20:02:58 Kryptonian kernel: [  117.987837] PeerBeaconAtScanAction P2P_SCANNING: DIRECT-64-HP OfficeJet Pro 8710 [1]
Aug 14 20:02:58 Kryptonian kernel: [  117.992344] PeerBeaconAtScanAction P2P_SCANNING: DIRECT-64-HP OfficeJet Pro 8710 [1]
[... cut ...]
Aug 14 20:03:08 Kryptonian NetworkManager[1434]: <info>  [1534294988.2938] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Aug 14 20:03:08 Kryptonian NetworkManager[1434]: <info>  [1534294988.2943] manager: NetworkManager state is now CONNECTING
Aug 14 20:03:08 Kryptonian kernel: [  128.182557] CFG80211DRV_IoctlHandle: CMD_RTPRIV_IOCTL_80211_NETDEV_EVENT
Aug 14 20:03:08 Kryptonian kernel: [  128.182563] BUG: unable to handle kernel paging request at 0000000e0012d8fd
Aug 14 20:03:08 Kryptonian kernel: [  128.182568] PGD 0 P4D 0 
Aug 14 20:03:08 Kryptonian kernel: [  128.182572] Oops: 0000 [#1] SMP PTI`

dabonde, 2018-09-06:
@kaduke Your fork fixed this problem for me on ubuntu 18.04. Thank you!

rdockterjr, 2018-09-24:
@kaduke fork worked for me as well. Is this a PR yet?