jurobystricky/Netgear-A6210 #121: make issue

RustyBalboadev, 2018-10-07:
Hi! When I use the command make inside the directory Netgear-A6210
I get this message:
root@kali:~/Netgear-A6210# make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /root/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.17.0-kali1-amd64/build DBGFLAGS=-DDBG SUBDIRS=/root/Netgear-A6210/os/linux modules
make[1]: *** /lib/modules/4.17.0-kali1-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:64: debug] Error 2

I used to have 2 errors but I fixed one by skimming through other reports and creating a file
i've ran so far:
apt-get install linux-headers-$(uname -r) and I get this
root@kali:~/Netgear-A6210# apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.17.0-kali1-amd64
E: Couldn't find any package by glob 'linux-headers-4.17.0-kali1-amd64'
E: Couldn't find any package by regex 'linux-headers-4.17.0-kali1-amd64'

I've also tried running:
sudo apt-get install kernel-headers-$(uname-r) and I get
root@kali:~/Netgear-A6210# sudo apt-get install kernel-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package kernel-headers-4.17.0-kali1-amd64
E: Couldn't find any package by glob 'kernel-headers-4.17.0-kali1-amd64'
E: Couldn't find any package by regex 'kernel-headers-4.17.0-kali1-amd64'

Thank you

ImConfused3, 2018-11-07:
I installed this on one computer with Linux Mint 18.3 Kernel 4.15.0-33. In this computer I have Linux Mint 19,1 Kernel 4.15.0-38. I don't know if I am in the right place to post but make is giving me:
~/Netgear-A6210$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/michael/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.15.0-38-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/michael/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-38-generic'
  CC [M]  /home/michael/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9:0,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/michael/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/michael/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/michael/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/michael/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/michael/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:340:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
scripts/Makefile.build:332: recipe for target '/home/michael/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/home/michael/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1551: recipe for target '_module_/home/michael/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/michael/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-38-generic'
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2

Dekken, 2019-01-29:
Seeing this on kernel 4.19 also

```In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/philix/git/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
/usr/src/linux-headers-4.19.0-0.bpo.1-common/include/linux/string.h:341:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
```

closed 2020-10-05 by RustyBalboadev