jurobystricky/Netgear-A6210 #122: 4.18.16

CGDoctor8, 2018-10-28:
I have been using @kaduke [port-to-4.15](https://github.com/kaduke/Netgear-A6210/tree/port-to-4.15/) but recently broke for 4.18 for me
`doctor@doctor ~/Documents/Netgear-A6210 $ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/doctor/Documents/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.18.16-96.current/build DBGFLAGS=-DDBG SUBDIRS=/home/doctor/Documents/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.16-96.current'
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/assoc.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/auth.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/auth_rsp.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/sync.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/sanity.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/rtmp_data.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/connect.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/wpa.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/sta_cfg.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../sta/sta.o
  CC [M]  /home/doctor/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/doctor/Documents/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/doctor/Documents/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/doctor/Documents/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/doctor/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/doctor/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:341:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:318: /home/doctor/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[1]: *** [Makefile:1505: _module_/home/doctor/Documents/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.18.16-96.current'
make: *** [Makefile:64: debug] Error 2
`

Mathisca, 2018-11-04:
Same issue, I had to downgrade Linux as a workaround

jafesellers, 2018-11-13:
I am having the same issue running 4.18.0
any ideas on how to fix it ?

Dovahki, 2018-12-04:
@Mathisca Can you please tell me how did you downgrade it? I searched the web but could not find a tutorial. Sorry I am a beginner in Kali Linux. 
Thank you

PranavMadanahalli, 2019-01-07:
@Dovahki did you ever successfully downgrade?

Dovahki, 2019-01-07:
@PranavMadanahalli I could not whatsoever. However, I bought another adapter which costed me roughly 7 US dollars. It works perfectly and I did not have to even download or install anything. Just plug and play.

Proximyst, 2019-06-29:
Note, you do not need to downgrade or buy some Edimax adapter. Linux v5 has native support for the A6210's chipset, so you only need to stay up-to-date :)

krdeepak, 2019-08-05:
Did someone get it working for latest ubuntu 18.04.2 which has kernel `4.18.0-15-generic` ?