jurobystricky/Netgear-A6210 #128: Make error

DukeOfJamming, 2019-04-25:
Hi, I am using Manjaro with the 4.19.34-1 Kernel installed. When I run the make file I get the following error:

[benjamin@benjamin-pc Netgear-A6210]$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/benjamin/Documents/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.19.28-1-MANJARO/build DBGFLAGS=-DDBG SUBDIRS=/home/benjamin/Documents/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.19.28-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.19.28-1-MANJARO/build'
make: *** [Makefile:64: debug] Error 2

I looked around at other people who had this problem. I have checked that I have the linux419 headers installed in package manager. I also tried making that folder since my modules folder only had a 4.19.34-1-MANJARO folder. The last thing I tried was manually setting the make file to use the folder that the make error was trying to use, that offered no change. When I set the make file to manually use the newer .34 folder it got a bit further and then got stuck saying the same error, the output is below:

[benjamin@benjamin-pc Netgear-A6210]$ make
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/benjamin/Documents/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.19.34-1-MANJARO/build DBGFLAGS=-DDBG SUBDIRS=/home/benjamin/Documents/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.19.34-1-MANJARO/build'
  CC [M]  /home/benjamin/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/benjamin/Documents/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/benjamin/Documents/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/benjamin/Documents/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/benjamin/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/benjamin/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:341:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:304: /home/benjamin/Documents/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[1]: *** [Makefile:1525: _module_/home/benjamin/Documents/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.34-1-MANJARO/build'
make: *** [Makefile:64: debug] Error 2

Since a bunch of old posts said you need the linux49 headers I tried installing them but from what I have read now I don't think I need them as that isn't my kernel version? Sorry I am kind of new to this so any help would be appreciated.

kasaurov, 2019-06-20:
The same Error on Ubuntu 18.04, Linux 4.15.0-53-generic #57-Ubuntu SMP Thu Jun 13 09:24:30 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

Proximyst, 2019-06-29:
Try updating to Linux 5(.1/.2); the newer kernel version has native support for the chipset the Netgear A6210 runs, per my testing on Arch Linux and Linux v5.1.12 (yes, I know .15 is newest, but oh well).

Azmisov, 2020-04-08:
Try this fork, which seems better maintained: https://github.com/Netgear-A6210-linux-driver/Netgear-A6210