jurobystricky/Netgear-A6210 #131: Make error

Nitr0f0x, 2019-07-21:
Hi, so I'm quite new to the linux community and I'm having issues making the driver I'm using the latest version of manjaro KDE edition and this is what the debug message comes up with:

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/nitr0f0x/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/4.19.59-1-MANJARO/build DBGFLAGS=-DDBG SUBDIRS=/home/nitr0f0x/Netgear-A6210/os/linux modules
make[1]: Entering directory '/usr/lib/modules/4.19.59-1-MANJARO/build'
  CC [M]  /home/nitr0f0x/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
In file included from ./include/linux/bitmap.h:9,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:21,
                 from ./arch/x86/include/asm/cpufeature.h:5,
                 from ./arch/x86/include/asm/thread_info.h:53,
                 from ./include/linux/thread_info.h:38,
                 from ./arch/x86/include/asm/preempt.h:7,
                 from ./include/linux/preempt.h:81,
                 from ./include/linux/spinlock.h:51,
                 from ./include/linux/seqlock.h:36,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:10,
                 from /home/nitr0f0x/Netgear-A6210/include/os/rt_linux.h:14,
                 from /home/nitr0f0x/Netgear-A6210/include/rtmp_os.h:30,
                 from /home/nitr0f0x/Netgear-A6210/include/rtmp_comm.h:64,
                 from /home/nitr0f0x/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:33:
In function ‘memcpy’,
    inlined from ‘rt_ioctl_iwaplist’ at /home/nitr0f0x/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:549:2:
./include/linux/string.h:344:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyon
  344 |    __read_overflow2();
      |    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:304: /home/nitr0f0x/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
make[1]: *** [Makefile:1517: _module_/home/nitr0f0x/Netgear-A6210/os/linux] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.59-1-MANJARO/build'
make: *** [Makefile:64: debug] Error 2

I was wondering if anyone is able to help me, I have updated manjaro packages to the latest but just having some trouble.

Thanks all for any help you can give me

florinbarbisch, 2019-09-21:
same issue here

florinbarbisch, 2019-09-21:
references #128

closed 2021-02-06 by xXD3ADSH00TXx