jurobystricky/Netgear-A6210 #132: Please,Help!! After install,It DOES NOT WORK ON Jetson Nano ARMx64 Ubuntu18.04

andersling, 2019-12-18:
Hi,all:
   I installed the driver by following  the instructions given by jurobystricky. During the installation process , i got a warning message:

  
[image]

and after i installed it , i plug in the Adapter A6210, it DOES NOT work at all,not event show in the iwconfig. 
  and i view the dmesg log:
  
[image]

[image]

ANY idea? Thanks a lot!!!!!

@jurobystricky

cagnulein, 2020-01-28:
@andersling i've had a similar issue here with a raspy 3 http://robertoviola.cloud/2020/01/28/netgear-a6210-ac1200-wifi-usb-adapter-on-raspberry-4-raspbian/

check at the bottom of the page, i've enlarged the kernel memory size. I think there is some tweak also in the jetson nano.

guation, 2021-02-07:
> @andersling i've had a similar issue here with a raspy 3 http://robertoviola.cloud/2020/01/28/netgear-a6210-ac1200-wifi-usb-adapter-on-raspberry-4-raspbian/
> 
> check at the bottom of the page, i've enlarged the kernel memory size. I think there is some tweak also in the jetson nano.

The 5.10.5-v8+ kernel of raspberry pi 4B can't make.

```
pi@PiRoute:~/Netgear-A6210 $ uname -a
Linux PiRoute 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
pi@PiRoute:~/Netgear-A6210 $ sudo make -j4
export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/pi/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/5.10.5-v8+/build DBGFLAGS=-DDBG M=/home/pi/Netgear-A6210/os/linux modules
make[1]: 进入目录“/usr/src/linux-headers-5.10.5-v8+”
  CC [M]  /home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o
  CC [M]  /home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_rx.o
  CC [M]  /home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_tx.o
  CC [M]  /home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211_inf.o
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: In function ‘CFG80211_OpsVirtualInfChg’:
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:343:47: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
   if (((*pFlags) & NL80211_MNTR_FLAG_FCSFAIL) == NL80211_MNTR_FLAG_PLCPFAIL)
                                               ^~
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:349:47: warning: bitwise comparison always evaluates to false [-Wtautological-compare]
   if (((*pFlags) & NL80211_MNTR_FLAG_CONTROL) == NL80211_MNTR_FLAG_OTHER_BSS)
                                               ^~
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c: At top level:
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2505:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = CFG80211_OpsMgmtFrameRegister,
   ^~~~~~~~~~~~~~~~~~~
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2505:25: error: initialization of ‘int (*)(struct wiphy *, struct net_device *, s32,  s32)’ {aka ‘int (*)(struct wiphy *, struct net_device *, int,  int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_dev *, u16,  bool)’ {aka ‘void (*)(struct wiphy *, struct wireless_dev *, short unsigned int,  _Bool)’} [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = CFG80211_OpsMgmtFrameRegister,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.c:2505:25: note: (near initialization for ‘CFG80211_Ops.set_cqm_rssi_range_config’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279：/home/pi/Netgear-A6210/os/linux/../../os/linux/cfg80211/cfg80211.o] 错误 1
make[2]: *** 正在等待未完成的任务....
make[1]: *** [Makefile:1808：/home/pi/Netgear-A6210/os/linux] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.10.5-v8+”
make: *** [Makefile:65：debug] 错误 2
```