jurobystricky/Netgear-A6210 #135: New kernel 5.3 build error:"access_ok" passed 3 arguments, but takes just 2

DanielJefferson1990, 2020-03-21:
`export DBGFLAGS

*** Building driver with debug messages ***

cp -f os/linux/Makefile.6 /home/user/Netgear-A6210/os/linux/Makefile
make -C /lib/modules/5.3.0-43-generic/build DBGFLAGS=-DDBG SUBDIRS=/home/user/Netgear-A6210/os/linux modules
make[1]: enter“/usr/src/linux-headers-5.3.0-43-generic”
Makefile:227: ================= WARNING ================
Makefile:228: 'SUBDIRS' will be removed after Linux 5.3
Makefile:229: 
Makefile:230: If you are building an individual subdirectory
Makefile:231: in the kernel tree, you can do like this:
Makefile:232: $ make path/to/dir/you/want/to/build/
Makefile:233: (Do not forget the trailing slash)
Makefile:234: 
Makefile:235: If you are building an external module,
Makefile:236: Please use 'M=' or 'KBUILD_EXTMOD' instead
Makefile:237: ==========================================
  CC [M]  /home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o
/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c: In function ‘rt28xx_sta_ioctl’:
/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:2205:73: error: macro "access_ok" passed 3 arguments, but takes just 2
   if (access_ok(VERIFY_READ, wrqin->u.data.pointer, wrqin->u.data.length) != TRUE)
                                                                         ^
/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:2205:7: error: ‘access_ok’ undeclared (first use in this function)
   if (access_ok(VERIFY_READ, wrqin->u.data.pointer, wrqin->u.data.length) != TRUE)
       ^~~~~~~~~
/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.c:2205:7: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:288: recipe for target '/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o' failed
make[2]: *** [/home/user/Netgear-A6210/os/linux/../../os/linux/sta_ioctl.o] Error 1
Makefile:1655: recipe for target '_module_/home/user/Netgear-A6210/os/linux' failed
make[1]: *** [_module_/home/user/Netgear-A6210/os/linux] Error 2
make[1]: leave“/usr/src/linux-headers-5.3.0-43-generic”
Makefile:59: recipe for target 'debug' failed
make: *** [debug] Error 2`

uname  -a
Linux deepmind 5.3.0-43-generic #36~18.04.2-Ubuntu SMP Thu Mar 19 16:03:35 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

how to fix it? it seems that kernel headers are updated

timmay237, 2020-04-02:
I'm also having this same issue

JasonShiao, 2022-03-04:
The Linux kernel API has changed a lot and this repo doesn't handle that.